jlab-sensing/ENTS-node-hardware #55: Support for battery charging

jmadden173, 2024-09-20:
While brainstorming the waterproof enclosure for the board, we realized that if the device is buried underground and the batteries run out of charge, the enclosure would need to be dug up and batteries replaced. Battery charging via an external connection fix this issue.

Brainstorming is required to determine the type of battery (stick with AA, or other form factor) and how we are going to integrate the charging circuit. I think there will be some difficulty

jmadden173, 2024-10-02:
Proof there is NiMH battery charging ICs that can be integrated on the board.
https://www.digikey.com/en/products/detail/texas-instruments/BQ25172DSGR/16585660

jmadden173, 2024-10-08:
Wikipedia has a good reference for comparison between battery technologies available [here](https://en.wikipedia.org/wiki/Comparison_of_commercial_battery_types#cite_note-OEM-33). The Enloop AA batteries are considered *low self discharge* making it comparable to LiPo and Li-ion. There's also [this](https://openenergymonitor.blogspot.com/2013/10/aa-battery-considerations.html) website that tracked the discharge rate of multiple battery types.

Unless there are any objections, I think we should stick to NiMH as already planned as we won't risk issues with moving to a 3.7V battery that can cause unforeseen side effects.

I listed the chip above to see if a multi-cell charger exists. Usually we prefer using ICs with non-QFN style pads (pads underneath the package) so that we can inspect the solder connections, so I would look around for other chips. Bonus points if they are available and in-stock on [JLCPCB parts](https://jlcpcb.com/parts).

jlin143, 2024-10-08:
Need some requirements on:
1. Size (volume)
2. Capacity and/or minimum lifetime (with charge, without charge)
3. Voltage (min, max, ideal/nominal)
4. Current discharge rate (min, max, average)
5. Weight
6. Operational temperature

jlin143, 2024-10-09:
[image]

3D render of a draft of the NiMH battery module based on the BQ25172 NiMH BMS. This model does not include solar MPPT or separate power path (i.e. battery and Vout are shorted). Will upload the design files to hardware shortly. To do:

- Fix DRC errors
	- Silkscreen overlaps
	- JLCPCB constraints (min clearance, etc)
	- Ground plane fill minimum spokes on USB-C connector through-holes.
- Select a 4-slot AAA tray instead of 2x 2-slot AAA trays.
- Reduce the number of input connectors to <= 2.
	- Currently have 2.1mm barrel, screw terminal, USB-C, micro USB.
- Consider moving from through-hole resistors and capacitors to SMD.
	- This does mean that the ease of modification is reduced.
- Move battery trays to underside. Requires less through-hole components to be used.
- Standoff and mounting holes, to be coordinated with the node mainboard and enclosure limitations.
- Consider board-to-board pin header connection (ex. arduino shield-esque) instead of screw terminal output.
- Source an equivalent thermistor (Semitech 103AT-2) on digikey. The recommended thermistor is not in stock on digikey.

jmadden173, 2024-11-12:
Reference to pinout
[image]

jlin143, 2024-11-13:
Updated schematic and PCB with part selection info and adjusted the battery cell footprint in commit 15fccb8429cf77c0dc20af99edd72b60c186cc3b. Should be ready for order / pull request. I recommend a small batch to verify the battery clip placement and spacing. The datasheet did not provide much information on the bend and stiffness of the battery tab, and despite the common name of 18650, cells can vary between 65mm and 70mm.

For the battery selection, I recommend one of the following (in order from cheap to expensive at 50qty):
1. [66.92mm] https://www.18650batterystore.com/products/samsung-35e-button-top?variant=37145681690775
2. [65.25mm] https://www.18650batterystore.com/collections/18650-batteries/products/samsung-35e
3. [65.05mm] https://www.18650batterystore.com/collections/18650-batteries/products/molicel-m35a-18650-battery

jmadden173, 2024-12-06:
Notes from design review of ents before large order (2024-12-05)
- Tested supplying 3V3 to the output of the on-board LDO on the Wio-e5. Experimentally validated that the 60 ohm discharge resistor is not in effect when the output pin is powered from a external 3V3 rail.
- Plan is to add a physical switch on the ents board that switches between USB and Battery. Only the `Vcc` rail switch is needed since the regulated 3V3 rails (Wio-e5, battery, circuit 3V3) will all be connected together. The Wio-E5 3V3 rails acts as both a power input and output.
- Updated the battery board pinout in the following order
    1. GND
    2. VBAT
    3. 3V3
    4. PG
    5. CHG
    6. CHG
    7. PG
    8. 3V3
    9. VBAT
    10. GND

Changed needed on *battery*:
- [ ] Add LDO to battery board
- [ ] Update pinout
- [ ] Update switch to one available on JLCPCB

Changes needed for *mainboard*

- [ ] Update pinout
- [ ] Added switch for power
- [ ] Implement the login shown in the following image:

[image]

commit fffd798e (2024-12-08): Merge branch 'battery_module' into release/2.2.3

Closes #55
PR #63 (2024-12-08): Release 2.2.3
## Task List

- [x] Version bump files
- [x] Update changelog
- [x] Export plots of schematic
- [x] Export BOM
- [x] Update BOM on Octopart
- [x] Generate fab files

## Relevant Issues

List issues related to PR
Closes #60 
Closes #59 
Closes #58 
Closes #57 
Closes #55
closed 2024-12-08 by jmadden173